bullitt186/ha-stiebel-control #1: Change room temperature?

robot300795, 2023-08-16:
Is it possible to change the room temperature for the heat pump which is sent for example from the FEK and is important for cooling? I would like to send the heat pump the lowest temperature of all rooms.

bullitt186, 2023-08-16:
Not out of the box, currently.
there are several signals such as RAUMISTTEMP, RAUMSOLLTEMP, RAUMSOLLTEMP_NACHT and VERSTELLTE_RAUMSOLLTEMP, so it should be possible in theory. However on my 2023 WPL13e (without cooling function) and a simple FE7 instead of a more advanced FEK, I only got -255 (SNA) for RAUMISTTEMP, so I cannot test this out properly.
I see the issue if we send out RAUMISTTEMP via Home Assistant, it would be overwritten by the SNA RAUMISTTEMP on the CAN Bus. Maybe one could deactivate the FEK completely but I have not tested this. 

If you want to test yourself: To add the writing capabilities to the raumtemperatures, you simply have to add entries in the ha_stiebel_control.yaml, such as:
```YAML
input_number:
  speichersolltemp2_set:
    name: "SPEICHERSOLLTEMP2_SET"
    initial: 20
    min: 20
    max: 60
    mode: box
    step: 1
    unit_of_measurement: "°C"
    icon: mdi:thermometer-low
```
and in the esphome sketch:
```yaml
  - platform: homeassistant
    name: "EINSTELL_SPEICHERSOLLTEMP2_SET"
    id: EINSTELL_SPEICHERSOLLTEMP2_SET
    entity_id: input_number.speichersolltemp2_set
    on_value:
      then:
        lambda: |-
          if(atoi(x.c_str()) != 20) {
            char temp[3];
            strncpy(temp, x.c_str(), sizeof(temp)-1);
            temp[2] ='\0';
            const char* ctemp = temp;
            writeSignal(&CanMembers[cm_manager], GetElsterIndex("EINSTELL_SPEICHERSOLLTEMP2"), ctemp);
            id(EINSTELL_SPEICHERSOLLTEMP2).update();
            id(SPEICHERISTTEMP).update();
            id(SPEICHERSOLLTEMP).update();
          }
          return;
```

In case you do so, let me know the results of your investigation

robot300795, 2023-08-17:
Good to know, I am also working on a controller for WPF13 with FEK and ISGWEB and came across your work. I wanted to create an ESp32 board connected via TCP Modbus. I can say that the FEK simply sends all available addresses it has to the bus. Among them also the room temperature and room humidity. Interesting is that the FEK sends every 60 seconds the room temperature with a value of 32768 and directly after (20-100 ms) the actual temperature.

The problem with the override I had also considered. My ideas was a kind of CAN bridge. All messages from the heat pump side are passed to the FEK, and values from the FEK are changed. The problem is that I currently still have an electrical problem. A Can Bus should have two resistors a 120 Ohm, but apparently Stiebel does not adhere to this and the CAN boards do not like that at all.

[timestamp, delta_time (sec), data]

395348 59.27 d2 00 fa 00 11 80 00
395348 59.27 [210, 0, 250, 0, 17, 128, 0]
395348 59.27 0x680 2 17 3276.8
-----------------------------
395514 0.17 d2 00 fa 00 11 01 14
395514 0.17 [210, 0, 250, 0, 17, 1, 20]
395514 0.17 0x680 2 17 27.6
-----------------------------
455314 59.8 d2 00 fa 00 11 80 00
455314 59.8 [210, 0, 250, 0, 17, 128, 0]
455314 59.8 0x680 2 17 3276.8
-----------------------------
455522 0.21 d2 00 fa 00 11 01 14
455522 0.21 [210, 0, 250, 0, 17, 1, 20]
455522 0.21 0x680 2 17 27.6
-----------------------------
515119 59.6 d2 00 fa 00 11 80 00
515119 59.6 [210, 0, 250, 0, 17, 128, 0]
515119 59.6 0x680 2 17 3276.8
-----------------------------
515528 0.41 d2 00 fa 00 11 01 14
515528 0.41 [210, 0, 250, 0, 17, 1, 20]
515528 0.41 0x680 2 17 27.6

My results and co I will also share

anilini, 2024-12-16:
> Not out of the box, currently. there are several signals such as RAUMISTTEMP, RAUMSOLLTEMP, RAUMSOLLTEMP_NACHT and VERSTELLTE_RAUMSOLLTEMP, so it should be possible in theory. However on my 2023 [...]

is it possible to create the input directly in the device, instead of using a helper...?