bullitt186/ha-stiebel-control #2: WPC-07-Cool

m2teusz, 2023-09-02:
Hi, 
I have a WPC-07 pump with the following modules: WPM3i, FES, FEK, MFG, unfortunately ESPHome (2023.8.2) does not read most of the sensors. Is it possible to adapt the code to my pump? I tested the code from "github.com/roberreiter/StiebelEltron-heatpump-over-esphome-can-bus" and here most of the sensors work but the control option is missing.  I used ESP-WROOM-32 + MCP2515 for communication. Attached is the log of your code as well as the code of colleague roberreiter.

[logs_esp-sensor-node-hzg_logs.txt](https://github.com/bullitt186/ha-stiebel-control/files/12504549/logs_esp-sensor-node-hzg_logs.txt)
[logs_heatingpump_logs.txt](https://github.com/bullitt186/ha-stiebel-control/files/12504550/logs_heatingpump_logs.txt)

bullitt186, 2023-09-03:
I am sailing this week, I'll have a look when I am back. 
Have you adapted the CAN IDs in the esphome sketch and heatpump.h as described in the readme?

m2teusz, 2023-09-04:
So I tried but with poor results, in addition, some addresses overlap and readings are missing.

mkaiser, 2023-09-04:
> So I tried but with poor results, in addition, some addresses overlap and readings are missing.

hi, 

I want to migrate from roberreiters stuff, too (tecalor TTF 07-cool, which is the same as the Stiebel eltron WPC-07cool) .

I did not have much time, yet for testing.  Only a few values (like temperatures) were working. 

@bullitt186 are you familiar with discord? We could join forces while debugging 

https://discord.com/invite/MkSF8qCn9Q

bullitt186, 2023-09-04:
Sounds good! I think one major feature to add would be some kind of "autoconfig", maybe by having lists of potential can member IDs and scanning which ones respond to e.g. a request of the sw version. 
I think this is one of the biggest hurdles for anybody setting this up (same for robreiters version)
I am sailing this week so I cannot do much this week

m2teusz, 2023-09-04:
@bullitt186 very cool idea! If testing is needed I will be happy to help when I return from vacation.

Monaco85, 2024-12-23:
are there any news of the CanID's for the WPC 07 Cool. I can just read a minor of them.