bullitt186/ha-stiebel-control #6: Add support for internal CAN controller

mkaiser, 2023-11-02:
hey there, 

in ESPHome 2023.10 I contributed some patches to support the internal CAN controller (our heatpumps need 20kbits) of newer ESP32 chips (see https://esphome.io/components/canbus.html?highlight=can) 

[image]

Although I have not managed to get all my heatingpump sensors to work, I want to share my results :)
( I guess it it still something with the different CAN addresses of my TTF-07cool, but it is definitvely a software-issue, because the results are the same when I use the mcp or the internal CAN)

I forked your project and inserted some comments meant as a template to use the internal CAN controller instead of the external SPI-connected mcp2515 (heatingpump.yaml).
Also, I renamed the can_controller id from "my_mcp2515" to the more generic "my_can_controller" in heatingpump.h

https://github.com/mkaiser/ha-stiebel-control/commit/954405d0b58c1cd6f3bd3a4d0a01b6dd261f25c7

When I uploaded to github, I did not notice, that my code beautifier messed up... 

So my questions here are:
- Would you accept such PRs in general?
- Could you define some code formatting, so merging would be easier?
- What do you think of refactoring the YAML-code a bit more, e.g., using the insertion operator?

https://esphome.io/guides/configuration-types.html#yaml-insertion-operator

BR

Martin

bullitt186, 2023-11-02:
Hey Martin, Thanks a lot for your contribution; I started some significant refactoring to allow a more modular approach but haven't checked in the changes yet. I'll review your PR in the next days and merge it with my changes. THanks a lot already!

mkaiser, 2023-11-02:
sounds good :)

let me know, if I can help / test something

mkaiser, 2023-12-31:
> Hey Martin, Thanks a lot for your contribution; I started some significant refactoring to allow a more modular approach but haven't checked in the changes yet. I'll review your PR in the next days and merge it with my changes. THanks a lot already!

hey,

I don't wont to push too much - but are there any updates?

btw. some of the HA community guys which participate often in the heatpump / can-related thread gathered here: 

https://discord.gg/y3Vp3RJTGq

If you join, we could maybe help a bit :)

cbialobos, 2024-08-30:
Hi, I'm using the integrated can controller:

canbus:
  - platform: esp32_can
    tx_pin: GPIO2
    rx_pin: GPIO1
    id: my_mcp2515
    can_id: 700
    use_extended_id: false
    bit_rate: 20kbps
    on_frame:

I'm using the m5stack S3 lite and the CAN bus Module, Grove