bullitt186/ha-stiebel-control #12: Support for WPE-I 08 HK 230 Premium

adrai, 2024-12-04:
I was able to run it with my WPE-I 08 HK 230 Premium, but I had to remove the `readSignal` intervals...
With the intervals, no message was received.
Also it seems only a few informations are processed correctly:

[image]

I suspect, I need a different ElsterTable... Do you know where we can find a documentation about that?

Here a debug log snippet:
```
[22:06:56][D][canbus:071]: received can message (#1) std can_id=0x500 size=7
[22:06:56][D][canbus:071]: received can message (#2) std can_id=0x700 size=7
[22:06:56][I][processCanMessage():115]: 700:	MODE MULTIFUNCTION OUTPUT:	346	(et_default)
[22:06:56][D][canbus:071]: received can message (#1) std can_id=0x500 size=7
[22:06:56][D][canbus:071]: received can message (#2) std can_id=0x700 size=7
[22:06:56][I][processCanMessage():115]: 700:	STATUS MULTIFUNCTION OUTPUT:	359	(et_default)
[22:06:56][D][canbus:071]: received can message (#1) std can_id=0x500 size=7
[22:06:56][D][canbus:071]: received can message (#2) std can_id=0x700 size=7
[22:06:56][I][processCanMessage():115]: 700:	INFOBLOCK_6:	39	(et_default)
[22:06:56][D][canbus:071]: received can message (#1) std can_id=0x500 size=7
[22:06:56][D][canbus:071]: received can message (#2) std can_id=0x700 size=7
[22:06:56][I][processCanMessage():115]: 700:	INDEX_NOT_FOUND:	208	(et_default)
[22:06:56][D][canbus:071]: received can message (#1) std can_id=0x180 size=7
[22:06:56][I][processCanMessage():115]: 180:	RETURN_FLOW_INTERNAL_TEMP:	-255	(et_dec_val)
[22:06:56][D][sensor:093]: 'RETURN_FLOW_INTERNAL_TEMP': Sending state -255.00000 °C with 1 decimals of accuracy
[22:06:56][D][canbus:071]: received can message (#2) std can_id=0x480 size=7
[22:06:56][I][processCanMessage():115]: 480:	RETURN_FLOW_INTERNAL_TEMP:	33.1	(et_dec_val)
[22:06:57][D][canbus:071]: received can message (#1) std can_id=0x500 size=7
[22:06:57][D][canbus:071]: received can message (#2) std can_id=0x700 size=7
[22:06:57][I][processCanMessage():115]: 700:	INDEX_NOT_FOUND:	70	(et_default)
[22:06:57][D][canbus:071]: received can message (#1) std can_id=0x500 size=7
[22:06:57][D][canbus:071]: received can message (#1) std can_id=0x480 size=7
[22:06:57][I][processCanMessage():115]: 480:	MODE MULTIFUNCTION OUTPUT:	0	(et_default)
[22:06:57][D][canbus:071]: received can message (#2) std can_id=0x700 size=7
[22:06:57][I][processCanMessage():115]: 700:	MODE MULTIFUNCTION OUTPUT:	346	(et_default)
[22:06:57][D][canbus:071]: received can message (#3) std can_id=0x480 size=7
[22:06:57][I][processCanMessage():115]: 480:	STATUS MULTIFUNCTION OUTPUT:	0	(et_default)
[22:06:57][D][canbus:071]: received can message (#4) std can_id=0x700 size=7
[22:06:57][I][processCanMessage():115]: 700:	STATUS MULTIFUNCTION OUTPUT:	359	(et_default)
[22:06:57][D][canbus:071]: received can message (#5) std can_id=0x480 size=7
[22:06:57][I][processCanMessage():115]: 480:	FLOW_TEMP_NHZ:	0.0	(et_dec_val)
[22:06:57][D][canbus:071]: received can message (#6) std can_id=0x480 size=7
[22:06:57][I][processCanMessage():115]: 480:	INFOBLOCK_6:	0	(et_default)
[22:06:57][D][canbus:071]: received can message (#7) std can_id=0x700 size=7
[22:06:57][I][processCanMessage():115]: 700:	FLOW_TEMP_NHZ:	36.3	(et_dec_val)
[22:06:57][D][canbus:071]: received can message (#8) std can_id=0x480 size=7
[22:06:57][I][processCanMessage():115]: 480:	INDEX_NOT_FOUND:	0	(et_default)
[22:06:57][D][canbus:071]: received can message (#9) std can_id=0x180 size=7
[22:06:57][I][processCanMessage():115]: 180:	RETURN_FLOW_INTERNAL_TEMP:	-255	(et_dec_val)
[22:06:57][D][sensor:093]: 'RETURN_FLOW_INTERNAL_TEMP': Sending state -255.00000 °C with 1 decimals of accuracy
[22:06:57][D][canbus:071]: received can message (#10) std can_id=0x480 size=7
[22:06:57][I][processCanMessage():115]: 480:	RETURN_FLOW_INTERNAL_TEMP:	33.5	(et_dec_val)
[22:06:57][W][component:237]: Component canbus took a long time for an operation (126 ms).
[22:06:57][W][component:238]: Components should block for at most 30 ms.
[22:06:58][D][canbus:071]: received can message (#1) std can_id=0x180 size=7
[22:06:58][I][processCanMessage():115]: 180:	RETURN_FLOW_INTERNAL_TEMP:	-255	(et_dec_val)
[22:06:58][D][sensor:093]: 'RETURN_FLOW_INTERNAL_TEMP': Sending state -255.00000 °C with 1 decimals of accuracy
[22:06:58][D][canbus:071]: received can message (#2) std can_id=0x480 size=7
[22:06:58][I][processCanMessage():115]: 480:	RETURN_FLOW_INTERNAL_TEMP:	33.1	(et_dec_val)
[22:06:59][D][canbus:071]: received can message (#1) std can_id=0x480 size=7
[22:06:59][I][processCanMessage():115]: 480:	MIXER_RUNNING_TIME:	0	(et_default)
[22:06:59][D][canbus:071]: received can message (#1) std can_id=0x480 size=7
[22:06:59][I][processCanMessage():115]: 480:	INDEX_NOT_FOUND:	0	(et_default)
[22:06:59][D][canbus:071]: received can message (#2) std can_id=0x700 size=7
[22:06:59][I][processCanMessage():115]: 700:	INDEX_NOT_FOUND:	20	(et_default)
[22:06:59][D][canbus:071]: received can message (#3) std can_id=0x180 size=7
[22:06:59][I][processCanMessage():115]: 180:	RETURN_FLOW_INTERNAL_TEMP:	-255	(et_dec_val)
[22:06:59][D][sensor:093]: 'RETURN_FLOW_INTERNAL_TEMP': Sending state -255.00000 °C with 1 decimals of accuracy
[22:06:59][D][canbus:071]: received can message (#4) std can_id=0x480 size=7
[22:06:59][I][processCanMessage():115]: 480:	RETURN_FLOW_INTERNAL_TEMP:	33.5	(et_dec_val)
[... 123 lines cut ...]
[22:07:06][I][processCanMessage():115]: 480:	RETURN_FLOW_INTERNAL_TEMP:	33.1	(et_dec_val)
[22:07:07][D][canbus:071]: received can message (#1) std can_id=0x500 size=7
[22:07:07][D][canbus:071]: received can message (#1) std can_id=0x480 size=7
[22:07:07][I][processCanMessage():115]: 480:	MODE MULTIFUNCTION OUTPUT:	0	(et_default)
[22:07:07][D][canbus:071]: received can message (#2) std can_id=0x700 size=7
[22:07:07][I][processCanMessage():115]: 700:	MODE MULTIFUNCTION OUTPUT:	345	(et_default)
[22:07:07][D][canbus:071]: received can message (#3) std can_id=0x480 size=7
[22:07:07][I][processCanMessage():115]: 480:	STATUS MULTIFUNCTION OUTPUT:	0	(et_default)
[22:07:07][D][canbus:071]: received can message (#4) std can_id=0x700 size=7
[22:07:07][I][processCanMessage():115]: 700:	STATUS MULTIFUNCTION OUTPUT:	359	(et_default)
[22:07:07][D][canbus:071]: received can message (#5) std can_id=0x480 size=7
[22:07:07][I][processCanMessage():115]: 480:	FLOW_TEMP_NHZ:	0.0	(et_dec_val)
[22:07:07][D][canbus:071]: received can message (#6) std can_id=0x700 size=7
[22:07:07][I][processCanMessage():115]: 700:	FLOW_TEMP_NHZ:	36.2	(et_dec_val)
[22:07:07][D][canbus:071]: received can message (#7) std can_id=0x480 size=7
[22:07:07][I][processCanMessage():115]: 480:	INFOBLOCK_6:	0	(et_default)
[22:07:07][D][canbus:071]: received can message (#8) std can_id=0x480 size=7
[22:07:07][I][processCanMessage():115]: 480:	INDEX_NOT_FOUND:	0	(et_default)
[22:07:07][D][canbus:071]: received can message (#9) std can_id=0x700 size=7
[22:07:07][I][processCanMessage():115]: 700:	INFOBLOCK_6:	40	(et_default)
[22:07:07][D][canbus:071]: received can message (#10) std can_id=0x700 size=7
[22:07:07][I][processCanMessage():115]: 700:	INDEX_NOT_FOUND:	211	(et_default)
[22:07:07][W][component:237]: Component canbus took a long time for an operation (122 ms).
[22:07:07][W][component:238]: Components should block for at most 30 ms.
```

adrai, 2024-12-04:
Maybe @kr0ner you know more about that?

kr0ner, 2024-12-04:
wrong repo, but ok ^^ I've been talking to some guy on HA forum and the logs from his Stiebel Eltron WPL13 look familiar to yours.

adrai, 2024-12-04:
> How keen are you finding those out and upstreaming them?

I'm really new to all this... I have my home assistant green having configured a esp32-c6-zero wired to a SN65HVD230 CAN Board and that wired to my WPE-I...
Never worked with these hw pieces so far... I'm a softy ;-)
But if you guide me, I can try to give my best.

kr0ner, 2024-12-04:
I mean you receive can messages so you are done with the hardware part :D

cbialobos, 2025-01-08:
I've noticed that IDs changed with the english translation

DaveCamenisch, 2025-06-18:
@adrai Can I just ask if you have made any progress with your project? I have a WPE-I 12 HK 230 Premium, which will probably be identical to your model.

adrai, 2025-06-18:
Yes: https://github.com/kr0ner/OneESP32ToRuleThemAll/issues/101